gkasprow/Sensor-Box #10: Availability

jordens, 2018-03-04:
Where and when and what price?

gkasprow, 2018-03-04:
@jordens,  this was not meant as a commercial project.
I design it in my free time as a hobby project to keep some balance between work and life :)
These are simple things that simply work and give satisfaction.
However if you or someone else want to contribute, I can make a few more copies of the hardware.
I will produce it as a single multi-board project to keep the tooling cost low.
So far I managed to test critical functionalities and confirmed chipset selection using devkits. Essential parts of these boards (PoE, GPRS, Ethernet, CAN) were already verified in other projects.
When? That's good question. I started it 3 months ago and spend roughly one day per week on it. Within a month I plan to produce it. Then SW development will take yet another 2 months. Most of it will be based on verified Arduino or Cayenne libraries so it should be quick. In the summer I want to install it in my 2 houses and in my flat in Warsaw.

gkasprow, 2019-05-29:
The HW has just arrived.
[image]

jordens, 2019-05-29:
Nice! Did you have a chance to change the dimensions slightly as for the banker io modules?

gkasprow, 2019-05-29:
You discovered that issue after I sent it for production.

quynh-nn, 2019-05-30:
Great! How do these modules connect to each others? And how do these modules connect hardly with each others (I mean a vibrated protection)?

gkasprow, 2019-05-30:
they are connected by [edge connectors](https://pl.mouser.com/datasheet/2/40/OpenEndedCard%20Edge_00-9159-1125649.pdf) 
The modules are enclosed in [DIN shells](https://www.tme.eu/Document/90ae178fa75e7b415a1db063121a9557/DIN-72-11.pdf)

quynh-nn, 2019-05-31:
Perfect!

gkasprow, 2019-06-04:
@jordens I tried these modules with DIN enclosures from[ ELBAG ](https://www.tme.eu/Document/90ae178fa75e7b415a1db063121a9557/DIN-72-11.pdf)and they fit perfectly.
I also tried with enclosures from Italtronics and they need some force during insertion and indeed, could be 0.5mm shorter. Which enclosures did you use? Camden?

jordens, 2019-06-05:
Camden. Their drawings have the width of the slot at 72.4 mm.

jordens, 2019-06-05:
Maybe the polish/Italian have more accurate injection molds than the British...

jordens, 2019-06-07:
And the Banker IO boards that I got are actually 72.8 mm plus the break-off tab residue (~0.2 mm on each edge). The Camden slots are 72.4 mm in reality. So that's an issue with the boards.