gkasprow/Sensor-Box #13: soil moisture sensor

gkasprow, 2018-09-18:
One day the water distribution failed  - one of the tubes slipped off the filter terminal and plants were not watered for a few days. Another time, I shut down the main water valve and left home for one week. Plants hardly survived. This was my inspiration to build distributed humidity sensor.
I want to have following features:
- long range. At one place the garden is 1ha big, I want to have ability to monitor some plants far away from nearest building. So it cannot work over Bluetooth. I preserve some plants during the winter time in the cellar, there is no WiFi available, so standards like LoRa seem to be fine.
- I want to measure humidity, and not just condition that it is humid or dry. So certain accuracy is desired. I want to correlate individual plants condition with humidity in long time. I found an article which says that measurement frequency need to be more than 50MHz. With such frequency there is linear relationship between probe capacitance and humidity in the soil and the measurement does not depend strongly on solil type
- I want it to run on CR2032 or AAA battery more than one year. One measurement per hour is sufficient.

So i built simple circuit to check proof of principle.
The schematic is simple:
[image]
The generator built on the Schmidt inverter generates ~50MHz square wave which is fed to the divider that consists of the capacitance probe and R9. Resulting voltage is rectified and fed to the uPC ADC input.
The sensor is piece of FR4 with two long strips, covered by plastic sleeve. 
This is very simple but has some bottlenecks:
- the frequency varies a lot between gates so another board may oscillate at 60MHz and yet another at 40MHz.
- the temperature coefficient of diode can be essential because the output voltage difference is low, less than 400mV between dry sensor and immersed in water.
So next iteration needs to have compensation circuit which is formed by another dummy divider and detector. I needs to be driven from another negator to not influence the main one.
The range of ADC is 2.5V, so let's amplify the detector output a little.
The amplifier does not like the input voltages very close to zero, and diodes need to be polarised continuously so let's shift the voltage to half of VCC. So removal of C6 would do the job.
[image]

gkasprow, 2018-09-18:
Frequency stability can be improved by adding crystal generator. But popular low cost oscillators are 25, 48, 50 or 100MHz. And still they cost at least 1$ per piece. But we need plenty of that sensors.
We already have hex inverter IC, so let's try to build oscillator out of gates.
Here is example circuit 
[image] borrowed from [here](https://assemblergames.com/threads/mod-mega-drive-2-pal-60hz-color-fix-with-a-2-pin-crystal.49686/).
But still, there are not such crystals that run at 80MHz. We need to run ordinary crystal at third overtone.
This is example configuration that would do the job
[image]. I borrowed it from [here](http://www.beckwithelectronics.com/abracon/quartzan.htm)
In an overtone mode, an additional inductor LI and capacitance CC is required to select the 3rd-Overtone mode, while suppressing or rejecting the fundamental mode. 
The Lc tank may be located at either input or output of the inverter. However, the Lc tank at the out put is referred, because it helps to clean up all unwanted modes before signal goes through the crystal.
We cannot use Schmidt type negator, we need to have linearised inverter which acts as linear amplifier.
Finally we end up with something like this:
[image]
C13 and L2 are tuned at 81MHz, C5 and C12 give together 19pF which is close to 18pF required by the crystal.

jordens, 2018-09-18:
Or go the classic 555 route. There are a bunch of boards like that on aliexpress for a $.

gkasprow, 2018-09-18:
Yes, but they use very low frequency. And they have accuracy similar to the human finger :) . Some use resistive method which degrade over the time. I have very different soil types around my home so want to measure it "properly". I know that this can be done easier, but FDR probe gives quite good results. I even purchased 2 of [them](https://www.tindie.com/products/pinotech/soilwatch-10-soil-moisture-sensor/) and I'm currently testing it. I like its shape and accuracy and want to build something similar. It inspired me. The circuit is molded in plastic but I assume it works in similar manner as the circuit above. I will calibrate it with various soil types and later on maybe write some paper on my research :)

gkasprow, 2018-09-18:
And here is preliminary drawing of the device implementing the features above:
[image]
[image]

To lower the cost, helical antenna is also integrated with the PCB

gkasprow, 2018-09-18:
And here is the proof of the very first concept with single detector circuit;
[image]

muz0k, 2020-04-18:
Hi Sir,
After two years later, I decide to make one for myself but can not decide if copying the same circuit you build or adding/changing some components with my very limited knowledge of electronics. I want to ask some questions. I just this circuit be more sensitive. If I could ever complete it, it will measure the moisture of hazelnut by sticking two needles(for me they are probes) inside a hazelnut.(I can't find any hazelnut moisture meter, there are meters for almost all other grains.)
First: Why didn't you add a resistor (Rs) before crystal? I don't know it is a must but it is said to be lifesaver of crystal.
Second: Instead of 27mhz crystal, can I use bigger ones like 30 or 40/50? Does it make circuit better or not?
Third: Can I use 74hc04 instead of lvc04? If yes, do I need rearrangement of components?(I hope not)
Last one: How about adding MCP3421(18bit adc with I2C)? To make measurement more sensitive(most probably the microcontroller I will use esp32, so much so good)
I am really looking forward to hear you as soon as possible.
Thanks in advance.
[image]

gkasprow, 2020-04-18:
> First: Why didn't you add a resistor (Rs) before crystal? I don't know it is a must but it is said to be lifesaver of crystal.

The logic gate already has a resistance which is sufficient in this case

> Second: Instead of 27mhz crystal, can I use bigger ones like 30 or 40/50? Does it make circuit better or not?

Actually, there are no 50Mhz crystals. These are ones for smaller frequency that run on usually third overtone. That’s why you need LC circuit. This circuit does not work on 27MHz but on 3*27Mhz

>Third: Can I use 74hc04 instead of lvc04? If yes, do I need rearrangement of components?(I hope not)

what's wrong with HC04?
I didn't try, but this should work

>Last one: How about adding MCP3421(18bit adc with I2C)? To make measurement more sensitive(most probably the microcontroller I will use esp32, so much so good)

This circuit won't benefit anything from 18bit ADC. Temperature stability and the method won't give you accuracy better than a few %. However if you need to detect subtle changes in humidity increasing ADC resolution may help, but don't expect anything better than noise.

muz0k, 2020-04-19:
Dear sir,
I really thank you for your time and fast answers. I am on adc noice suppression research.

ikotzig, 2020-08-09:
Dear Greg,

My name is Ivan Kotzig, I am from Slovakia. I also want to make inexpensive, but above average sensor. I can invest into tooling, as I own a molding company.

Did you advanced further from the above schematics? I was considering a 100MHz circuit I found on Internet, but the components cost about 10 Eur, which is above my intentions.

I am a firmware developer, so I plan to ad an inexpensive CPU to your above schematics. I need 16 sensors connected by wire, not a long distance.

Any information would be welcome.
Ivan Kotzig
ivan<at>toolsquare<dot>com

[100MHz_sensor.pdf](https://github.com/gkasprow/Sensor-Box/files/5047943/100MHz_sensor.pdf)

gkasprow, 2020-08-15:
The entire design (sch and PCB) is already in the repo. I built the PCB, I also done some FW development but have other priorities right now.

petergzmo, 2021-02-21:
> The entire design (sch and PCB) is already in the repo. I built the PCB, I also done some FW development but have other priorities right now.

@gkasprow can you tell me where the sch and PCB are located in this repo?  I can't seem to find them...

gkasprow, 2021-02-23:
it's here
https://github.com/gkasprow/Sensor-Box/tree/master/Sensor_Box_Hardware/Sensor-Box/Sensor_Box_LoRa_Flower_Wachdog

ollyboy, 2022-01-11:
Your design is very good. Did you do any testing, would be great to see some real world results

gkasprow, 2022-01-11:
A batch was produced. I will install them once the spring comes :)

ollyboy, 2022-01-12:
if you use mcpl604 and the spare gates you could have 3 sensor circuits, top, middle, lower. This would provide better data and eliminate the possibility of misleading low capacitance paths in soil with organic materials and debris. Also maybe some gain on the output to get 0-3.3v range. Also why the 10k on output, would not <1k be better given micro controller ADC impedance.

ollyboy, 2022-01-12:
LT6905 oscillator may be better than xtal for single probe design, cheaper than the xtal,chip,inductor combo

gkasprow, 2022-01-12:
@ollyboy This design is good enough. One can make it much more precise, of course. But the question is what you want to achieve. I want to know if my plant is dying or not ;) It's not laboratory equipment. The 10k output resistance does not matter. The ADC S&H circuit needs low impedance and it's provided by a 1nF cap. The S&H circuit draws ns-scale current pulses from the source, if you leave just 1k resistor, it would be far too high. 1nF cap does the job nicely. LTC6905 is fine, but these days chips appear and disappear. It's easier to find replacement for classic crystal or logic gates.

delta77101, 2022-12-06:
Is there any further update on this design? Did you run any tests on your original set of boards? What range of voltages did you end up getting from dry to fully wet soil?

gkasprow, 2022-12-06:
Didn't do any other tests. I installed an automatic off-the-shelf irrigation system, so don't care about individual plants' moisture.

anilaras, 2024-11-19:
hello, I tried to make my own soil moisture sensor by modifying your circuit a little and using probes. However, when I examine the sens_out pin of the sensor with an oscilloscope, I see that there is a strange frequency wave here, not an analog voltage value. Is there a mistake in my design or is it supposed to be like this? How do I measure this value with an adc and make sense of it?
[image][image]

gkasprow, 2024-11-19:
what modifications did you make?
Where do you measure your signal? How do you connect the ground?
you get 20mV of noise, which may be acceptable, what is DC component?

gkasprow, 2024-11-19:
You need to adjust the generator resistor values to match your probes impedance otherwise you will saturate the amplifier

gkasprow, 2024-11-19:
the same applies to the capacitance

anilaras, 2024-11-19:
I know I'm asking too many questions, please excuse me. But I don't have much training in analog electronics. I wonder how I can measure the impedance of the probes and adjust the resistance and capacitor values ​​accordingly. Is it possible for you to point me to a source or a way where I can read and learn about this subject?

gkasprow, 2024-11-19:
just increase the resistor values, I'd start with 2x. Capacitance can be measured using DMM